tretyakovsa/Sonoff_WiFi_switch #150: Не работает кнопка

multysea, 2019-02-04:


multysea, 2019-02-04:
Залил стабильную версию прошивки в два новых чистых модуля. Не менял ничего, кроме того, что назвал модули и указал адреса. Версия - smart-room.  При нажатии на кнопки 1 и 2  все происходит штатно. При нажатии на кнопку 3 (gpio13) не происходит ничего. При этом светодиод подключенный к этому же выводу во время подключения к сети моргает, т.е. сам вывод работает точно. Ситуация полностью аналогична и на втором свежепрошитом модуле.  Подскажите, почему так происходит? Надеюсь, ошибка не в ДНК)).
[image]

renat2985, 2019-02-04:
Да 13 PIN занят процессом миргания при подключение к сети. Если вы хотите его повесить на кнопку, вам нужно зайти на странцу "Конфигурация" и выключить или поменять PIN здесь:
[image]