tretyakovsa/Sonoff_WiFi_switch #151: Предложение по активации режима точки доступа.

multysea, 2019-02-06:


multysea, 2019-02-06:
Может быть стоит сделать так, чтобы модуль не входил самостоятельно  в режим точки доступа после исчезновения питания или слишком долгого подъема роутера после восстановления питания? **Как вариант принудительного перевода в режим AP - использовать кнопку Reset, но не зажимать ее, чтобы не ввести модуль в режим программирования,  а сделать несколько нажатий с определенным интервалом, например 3-4 нажатия с интервалом  0.5-1.5 сек?**   Если модуль совсем без кнопок, можно было бы сделать принудительный перевод в режим точки доступа замыканием каких-нибудь контактов перед включением между собой или на шины земли или питания.

tretyakovsa, 2019-02-07:
@multysea Думаю предложение хорошее.

renat2985, 2019-02-08:
Мне больше нравится идея добавить возможность, отключить попытки подключатся ESP к роутеру.
[image]

И на странице сценариев добавить возможность включать АП. Тогда человек сам сможет настроить его включение на нужную ему кнопку и на нужную длительность нажатия.
[image]

multysea, 2019-02-08:
Можно и так. Хотя, если на плате есть хотят бы одна аппаратная кнопка (reset) то с точки зрения пользователя инициация режима стартового подключения к роутеру путем трехкратного нажатия кнопки выглядит проще и, к тому же, это  унифицировано для всех модулей. Но проект ваш - вам и решать:-).