tretyakovsa/Sonoff_WiFi_switch #153: Имитация нажатия кнопки

psvb28, 2019-02-25:
Приветствую всех участников данного сообщества!
Каким образом возможно организовать имитацию нажатия кнопки?

psvb28, 2019-02-25:
Добавлю немного конкретики, видимо решение моей проблемы все же в этом. Get запрос "http://ip/cmd?command=pulse on relay3 400" отрабатывает как и задумано, реле включается на 400мс (в данном случае) и отключается. Эта же команда через сценарий ЗАЦИКЛИВАЕТ процесс т.е. реле включается на 400 мс и отключается, включается и отключается и так по кругу.
Настройки производились на разных версиях прошивки, стабильной и бета - результат один.

renat2985, 2019-02-25:
Включить реле под номером 1
    http://ip/cmd?command=relay on 1 

 Выключить реле под номером 1
    http://ip/cmd?command=relay off 1

Переключить реле под номером 1
    http://ip/cmd?command=relay not 1 

Очень советую вам почитать WIKI https://github.com/tretyakovsa/Sonoff_WiFi_switch/wiki/GET-%D1%83%D0%BF%D1%80%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5

psvb28, 2019-02-25:
Благодарю за оперативность Ренат.
Опишу свою ситуацию, 
Задача: кратковременно  нажимать физическую кнопку (замыкать контакты) с помощью esp через нажатие кнопки в веб интерфейсе
Решение как вижу его я: на странице конфигурации включаю relay1 на несуществующем gpio25 (для отображения кнопки в веб) и pinout на gpio2 для управления. Добавляю сценарий на устройстве: ЕСЛИ "состояния реле1" равно 1, +или "состояния реле1" равно 0, ТОГДА на этом же устройстве "пульсация" on pinout1 400. Сохраняем, переходим на главную страницу устройства и нажимаем кнопку включить и наблюдаем: сценарий выполняется по кругу дёргая gpio2 бесконечно.
Отправив аналогичный get запрос из WIKI  "http://ip/cmd?command=pulse on pinout1 400" gpio2 дёргается единожды.
P.S.: не претендую на правильность логики выполнения задачи, наверняка есть более простое решение собственно с этим и обратился.
P.P.S.: сорри за много букав  :)

renat2985, 2019-02-25:
Звоните в Skype: renat2985

psvb28, 2019-02-25:
Как временное решение, Ренатом было предложено создание макроса, который отображается на главной странице и может быть использован в качестве кнопки. Однако работать будет только в последних бета прошивках

renat2985, 2019-02-25:
Сделал вот таких 2 сценария:

Первый:
Если Состояние кнопки 0 = 1 
Тогда  Управление реле not 1

Второй:
Если Состояние реле 1 = 1
Тогда Пульсация on Состояние реле 1 400 

Теперь нажимая на кнопку включения реше, оно включается на 400мс и выключается.

psvb28, 2019-02-25:
Вот оно! Отлично, то что надо!
Спасибо

closed 2019-02-25 by psvb28