tretyakovsa/Sonoff_WiFi_switch #155: Обновление прошивки

Forldol, 2019-03-15:
Здравствуйте уважаемый разработчики. Планируется ли обновление прошивки? Или проект заброшен?

renat2985, 2019-03-16:
Жив, жив :) пока заняты немного другим проектом. Но разработки этого тоже ведутся, просто пока не выкладывали на github. Хотим переехать на новое 2.5.0 ядро.

Forldol, 2019-04-17:
Стесняюсь спросить, а когда, хоть примерно, ждать новой версии? :)

renat2985, 2019-04-17:
В некоторых дизайнах есть вот такие блоки, там можно увидеть дату и фиксы. В тот же день обновляется и bin файл в бета версии. Которую можно скачать на странице Debug. Но установив бета версию нельзя переходить на стабильную (красный треугольник на главных страницах).
[image]

Forldol, 2019-04-18:
Понял, спасибо