tretyakovsa/Sonoff_WiFi_switch #156: Проблема с прошивкой

Forldol, 2019-04-17:
Здравствуйте, возникла такая проблема. Прошиваю ESP-07 последней прошивкой, всё прошивается нормально, появляется точка доступа, к ней подключается но после внесения любых изменение и перезагрузки устройства получаю вот такое

[image]

Что можете посоветовать?

renat2985, 2019-04-17:
В 90% у вас на ESP установлен Flash Chip который плохо работает с файловой системой. Зайдите на страницу http://192.168.*.*/page.htm?debug 
И покажите Flash Chip

Forldol, 2019-04-17:
[image]

renat2985, 2019-04-17:
Да, видите у вас красным написано Bad Chip. Вы не сможете нормально работать с файловой системой. Как только ESP8266 что-то попытается записать в файловую систему, сразу же сломает этот файл. Об это много писали у нас и много пишут в интернете. Вроде как новое ядро уже научили работать с этими Flash чипами, мы еще не проверяли.
Решения проблемы: Некоторые перепаивают Flash чипы. Или же ждите, возможно со временем все будет работать.

Forldol, 2019-04-17:
Это печально, подскажите какие чипы памяти туда можно запаять

tretyakovsa, 2019-04-19:
@Forldol Не торопитесь перепаивать. Нужно проверить будут ли работать ваши модули с новой версией. Скоро выложу новую бету.

Forldol, 2019-04-22:
@tretyakovsa Напишите пожалуйста сюда когда можно пробовать, я обязательно попробую =)

AntKul, 2019-08-19:
У меня такая же проблема с модулями ESP-01 из Китая. Ждемс новую версию.

Forldol, 2019-08-19:
Не дождался, перевел все свои модули на wifi-iot