tretyakovsa/Sonoff_WiFi_switch #160: IoT room Finder

Bender229, 2019-05-04:
Добрый вечер,хотел задать вопрос:настроил андройд прогу,но голосовые команды почему то не срабатывают,правда задействована 1 есп на ней управление реле.Возможно ли то что нужна база(есп) а от нее уже другие с устройствами,чтоб корректно все работало,или я не один такой?)
Скрины смысла показывать нет,так как ввел айпи устройства,команда по всплывающему окну показывает правильный код отправки.

tutaf, 2019-05-04:
@Bender229
Добрый. Покажите как задали команды на esp. Возможно где-то лишний пробел/точка/буквы не в нижнем регистре

Bender229, 2019-05-04:
[image]
вот так вот оформлял,даже с нижним подчеркиванием пробовал.

Bender229, 2019-05-04:
> 
> 
> @Bender229
> Добрый. Покажите как задали команды на esp. Возможно где-то лишний пробел/точка/буквы не в нижнем регистре

может нужны скобки или же  спец символы как в иногда требуется в IFTTT?

tutaf, 2019-05-04:
@Bender229  первую букву маленькой попробуйте сделать

Bender229, 2019-05-04:
> 
> 
> @Bender229 первую букву маленькой попробуйте сделать

окей сейчас испытаю

Bender229, 2019-05-04:
> 
> 
> > @Bender229 первую букву маленькой попробуйте сделать
> 
> окей сейчас испытаю

благодарю,да работает на нижнем регистре,буду знать)