tretyakovsa/Sonoff_WiFi_switch #164: IR передатчик SendRaw.

multiminik, 2019-06-12:
Есть проблема с пультом от ресивера Yamaha. Протокол передачи Nec. С частью кнопок проблем нет, т.к. их коды не выходят за пределы  максимально возможного положительного целого числа(0х7fffffff) для 32 битной переменной. Но есть кнопки с кодами, которые преодолевают этот предел в 2 раза (например: 0хFE80DA24). При отправке таких кодов с esp8266 всегда на выходе получим код - 0x7fffffff.
Отсюда вопрос? Есть ли в вашей прошивке возможность использовать SendRaw() из библиотеки irremote, не прибегая к ковыряниям в исходниках? Иначе проще будет подправить SendNec и перекомпмлировать прошивку.

multiminik, 2019-06-19:
Отвечу сам себе.
Функция SendRaw() не задействована в вашей прошивке. Пришлось всё же заглянуть в исходный код. Что касается предела в "0x7fffffff", - оказалось библиотека irremote не причём. Проблема была в 21 строчке скетча IR.ino. Необходимо в присвоении переменной "code" сменить функцию "strtol()" на "strtoul()". Эта замена позволит работать только с беззнаковыми целыми числами и расширит диапазон до заветных "0xffffffff".

tretyakovsa, 2019-06-20:
@multiminik Огромное спасибо за сообщение.