tretyakovsa/Sonoff_WiFi_switch #166: Поддержка Sonoff 4CH Pro

khseal, 2019-06-26:
Заявлено что прошивка поддерживает несколько реле. При установки на этот девайс можно настрость только одно реле и девайс определяется как Sonoff RF. 
Можно ли как-то настроить остальные реле или девайс так пока и не совместим?

khseal, 2019-06-26:
Отвечу сам себе. В режиме Manual все работает.
Вот скрин пинов реле
[image]
Настройки кнопок выложу позже...

khseal, 2019-06-26:
Ничего вроде не менял, чисто ковырялся с настройками сценариев кнопок. В итоге сломалось переключение реле. Странно.

renat2985, 2019-06-26:
Спасибо. Добавим пины в sonoff конфигурацию. Покажите какие вы сценарии создаёте. Все должно работать нормально.

khseal, 2019-06-26:
Я уже сменил прошивку.
Создавал сценарии на примере кнопки, которая была по умолчанию. Если GPIO0 = 1 то переключение реле.
В конфиге вот эти кнопки были прописаны
GPIO0	Button Channel 1	E-FW
GPIO9	Button Channel 2	IO9
GPIO10	Button Channel 3	IO10
GPIO14	Button Channel 4	IO14
В целом и RF модуль глючит т.к. как я понял он имитирует нажатие кнопок. GPIO9 и GPIO10 не работают совсем и при их добавлении начинаются странности.

multiminik, 2019-06-26:
Если у вас на странице http://х.х.х.х/page.htm?debug так, как на скрине ниже, то пины с 6 по 11 работать не будут. Если там будет DOUT, то должно работать.
[image]

khseal, 2019-06-26:
Прошивал DOUT было так же как на скрине, но gpio не работали =(