tretyakovsa/Sonoff_WiFi_switch #167: Сценарии

reo1978, 2019-07-06:
Здравствуйте подскажите как создать сценарии на одну кнопку управление несколькими реле: одно нажатие вкл 1-ое реле, два нажатия вкл. 2-ое реле, долгое нажатие вкл 3-е реле...увас написано что на одном пине можно сделать несколько кнопок с разной задержкой нажатия ..как это можно осуществить в сценариях?

renat2985, 2019-07-07:
Сценарий по двойному нажатию сделать нельзя.
Но можно сделать сценарии по продолжительности нажатия. Для этого в Конфигурации пинов сделайте несколько кнопок с разной задержкой на доном и том же пине.
Примерно так.
[image]

Где Tach 0 будет срабатывать сразу, Tach 1 при нажатие 1сек и Tack 2 при нажатие больше 2 секунд. Далее делайте сценарий на эти кнопки.

reo1978, 2019-07-07:
Да именно так я и пробовал делать....поэтому и прошу привести пример сценария  потому что у меня таким образом получилось только две кнопки  более менее стабильными ...а вот третья не получается помогите   составить сценарий по вашим соображениям на три кнопки ...подключение такое кнопки такоеже как у вас..в теме про подключение кнопок!? Заранее благодарен