tretyakovsa/Sonoff_WiFi_switch #168: Нужна ли полная версия?

nvash, 2019-07-06:
Скачал вашу прошивку - она замечательная , альтернатив нет :).
Но у меня не получается добавить новый блок на главный экран - не нашел такой возможности. Также не совсем понятно, как вывести показания температуры с ds18b20, и можно ли построить график температуры.
Есть ли эти возможности в полной версии или их нужно доделывать самому? 
При попытке обновиться(прошивка сама предлагает) проходит несколько минут и esp уходит в циклическую перезагрузку.

renat2985, 2019-07-07:
Добавление новых блоков (редакрирование страниц через CTRL+M) - доступно в полной версии.

Если вы хотите добавить график с тимпературой на главный экран вам необходимо зайди в Конфигурация \ Настройки разработчика \ Если вы работаете с NodeMCU или Wemos, желательно выбрать модуль Manually (там больше возможностей). И ставите галочку на ds18b20 и выбираете нужный вам PIN (менять их не рекомендуется).

[image]

В дизайнах Default и Life вы увидите такие графики:
[image]

nvash, 2019-07-07:
Если датчик не подключен - то графика не будет?

renat2985, 2019-07-07:
Да, если датчик не подключен - графика не будет. ds18b20 - вы можете подключать параллельно сколько угодно - графиков будет больше.

По поводу обновления, затрудняюсь ответить. Качайте архив с bin файлами и заливайте в свою ESP8266. Вы можете скачать бета (последнею) версию на странице "Конфигурация - Настройки разработчика - Debug". После того как вы установите бета версию на главной странице у вас будет появляться треугольник предлагая поставить снова стабильную версию - ее ставить нельзя, переход снова на стабильную версию мы еще не доделали.