tretyakovsa/Sonoff_WiFi_switch #172: Просьба и несколько советов

olegumar, 2019-11-30:
Здрасте всем.
Сразу хочу сказать, что я новичок в Ардуино, год как занимаюсь. Увидел уроки Третьякова, понравилось все. Сейчас делаю свой проект и решил взять за основу оболочку (платформу) из урока№8. Удобно тем, что хорошо просматривается на смартфоне, что не скажешь про весь проект который сейчас у вас есть. С HTML  знаком, но не спец, потому не стал брать где есть редактор страниц. Столкнулся с несколькими проблемами, что уже решил, но что то так и не могу.
Что хо посоветовать 
1. В комментариях добавьте номер версии библиотек на создания скетча. Так у меня появилась первая трудность. В уроке json 5 версии, а у меня стояла уже 6 и скетч не работал. Какая сейчас у вас стоит версии?
2. Прочитал много здесь тем. Ваш проект растет, это хорошо, но действительно скоро может получиться, что не влезет в ESP и тогда без карты SD не обойтись. Не понимаю зачем пихать все библиотеки и весь код, все датчики все равно не запихаете. Предлагаю разбить по вкладкам (что уже и так есть), каждый модуль и прикреплять только те вкладки (ino) что кому нужно. Заказ себе датчик BME280, его вроде нет в вашем списке...
3. Сам еще не столкнулся, но уже думал об возобновлении связи с роутером. В FiWi-менеджере есть сканер сети. Так вот, когда модуль переходить в точку доступа, надо переодически (прим раз в 1 мин) сканировать сеть. Как появилась ваша сеть, так и пытаться к ней подключиться. Сам еще не реализовал это, так как есть нюансы. Как роутер появился в сети, к нему не даст подключиться сразу, пока он полностью не загрузиться. Здесь были две мысли, предлагаю: как увидел - ставит метку истина, и следующую мин вместо скана пытаться подключиться.
Много что еще хото написать, но пока на этом все.
Есть просьба помочь запихнуть код вкладки NTP в урок№8 Третьякова.
А так же хотелось бы, что б Сергей продолжал тот проект как "чистую" оболочку без библиотек датчиков и с удобным просмотром на телефонах.

olegumar, 2019-12-12:
Все пробую закачать ваш проект, но ни как не получается, а так хото посмотреть все его возможности воочию.
У меня есть NodeMCU v3 (4M) (CH340G) на него и пробую.
Скачал архив Build.Spiffs.flash.1Mb.256Kb.2018.12.03.zip, октрыл NodeMCU Flasher, Нажал на Restore defaul.  Flash size показала 4М (думаю что так и должно быть). Далее выбираю оба файла и ... вопрос, какие адреса нужно ставить для моей NodeMCU? 
На скрине у вас стоит  build.0x00000, а для spiffs.0xBB000, но это для ESP  у которых 1М памяти.
С этими адресами у меня не видет FS, не грузиться ни чего. Пробовал spiffs.0x10000 (где то вычитал) то же не получилось. Подскажите как настроить Flasher?

tretyakovsa, 2019-12-12:
@olegumar  После того как скачали распаковали и запустили не нажимайте ни каких кнопок сделайте только выбор порта и затем сразу FLASH и ждите

olegumar, 2019-12-14:
Ни чего не получилось, вернул обратно свой проект.

Когда делал по своему, то появлялась сеть sonoff-loader (как то так). Заходил на .4.1 и видел WiFiменеджер, там находил свою сеть и вписывал данные. После перезагрузки ни где его не видел. Сейчас же, сделав как сказали, достал архив, сразу прошил и ... ни где ни чего ...
Пробовал грузить уроки 10 и 14, то же не смог, по разным причинам, в основном ругается на FS.h
Жду урок 15. Хотелось бы увидеть передачу данных с одной ESP на другую, когда все подключены к одному роутеру. 
Еще, хото скинуть благодарность, но у меня нет эл. кошельков, а перевод туда с карты просит большие проценты. Где посмотреть номер карты, что б на нее закинуть.?

tretyakovsa, 2019-12-14:
@olegumar Загрузив код указанным способом. Подождите 2 минуты и появится точка доступа. Подключайтесь к ней. 
По поводу благодарности со страницы https://esp8266-arduinoide.ru/thank/ вводите в форму сумму пишите текст выбираете заначек visa если отправляете с карты.  Произойдет переход на страницу яндекса вводите данные карты и процентов не будет.

olegumar, 2019-12-14:
Мой проект (на основе урока8) тоже не сразу появляется, я про это уже знаю. Ждал, курил и снова ждал...
"Спасибо" отправил, пробовал пока ... Сколько надо благодарностей за урок 15?

tretyakovsa, 2019-12-14:
@olegumar Спасибо за донат. Еще вы можете проверить после прошивки, а не подключен ли модуль уже к вашей сети. Так как предыдущие настройки сохраняются не зависимо от прошивки :)

olegumar, 2019-12-14:
К сожалению в пк нет WiFi, а роутер не всегда показывает (глючит) все подключичные устройства.

Возможно причина в самой ESP, вот тут заказывал [https://ru.aliexpress.com/item/32879758161.html?spm=a2g0s.9042311.0.0.274233edF9c4yi](url)
Она не похожа на те, что обычно рисуют на сайтах. С ней бывают какие то хлопаты, то все проходит хорошо, то что то не нравиться. Хотя возможно я сам что то еще не до понимаю.

tretyakovsa, 2019-12-14:
@olegumar Вам не нужен WIFI на компьютере если компьютер подключен проводом к роутеру.
Если под Windows загляните в папку Сеть. На андроид телефоне подключенном к вашему роутеру поставьте приложение https://play.google.com/store/apps/details?id=com.tutaf.iotroomfinder
Оно само все найдет.

olegumar, 2020-01-10:
еще раз Здрасте. Ни как не могу разобраться с Time.h (все взял из урока 8, изменил из урока 13)
 Вначале время из инета брало как подобает. Потом стало не всегда правильно показывать. Сейчас, заметил, выдает точку отчета, т е  1янв1970 00;00;00 Прибавляет зону и от нее тикает, и так при каждом запуске. Попробовал поставить NTPClient.h, то же самое ...
Ни где ни могу найти ответа, в чем может быть причина. Может подскажете.?
Возможно, когда стал добавлять свой код, и там что то мешает ...?
Мой весь скетч лежит https://github.com/olegumar/SmartClock