tretyakovsa/Sonoff_WiFi_switch #173: esptool.py + Linux

savegame, 2019-12-10:
Не подскажите как шить через esptool?  я вроде прошиваю, но точка доступа **Sonoff-Start** не появляется.
Команда которой прошиваю:
```
sudo esptool --port /dev/ttyUSB0 --baud 921600 write_flash --flash_size 1MB -fm dio -ff 40m 0x000000 build/build.0x00000_flash_size_1Mb.256Kb_2018.12.03.bin 0xbb000 build/spiffs.0xBB000_flash_size_1Mb.256Kb_2018.12.03.bin 
```
параметры **baud** провобал ставить 115200
параметр **--flash_size** пробовал ставить **detect**
параметр **-ff** пробовал менять на **keep** и прочее

сама прошивка идет без ошибок, далее, подключение ,переподлючение не дает никаких результатов ,точка доступа в сети не появляется.

savegame, 2019-12-10:
на крайняк попробую себе куда нибудь винду поставить... может через тулзу виндовую все продет гладко, отпишусь потом.
И вообще огромный респект ща прошивку!

renat2985, 2019-12-12:
Скорее не поможет. Но попробуйте с 0x000000 удалить 0. Должно быть так 0x00000

savegame, 2019-12-12:
Спасибо за ответ ,я попробую... 
Я в ESP новичек, и мне тут подсказали еще одну прошивку, и IDE PlatformIO, скажите ,ваш проект через PlatformIO компилится? было бы проще разобратсья в чем причина не работоспособности... )