tretyakovsa/Sonoff_WiFi_switch #177: Вопрос по MQTT

klepan63, 2020-01-26:
Приветствую
прошил Sonoff Basic
не могу найти где прописывать настройки MQTT 
в настройках ПИН выбрал MQTT но на странице конфигурации настройки не появляются

renat2985, 2020-01-26:
Читайте wiki этого репозитория.

klepan63, 2020-01-27:
если Вы про эту статью: https://github.com/tretyakovsa/Sonoff_WiFi_switch/wiki/MQTT-%D1%83%D0%BF%D1%80%D0%B0%D0%B2%D0%BB%D0%B5%D0%BD%D0%B8%D0%B5
то я ее читал НО
мне не куда вписывать настройки MQTT т.к. на странице "Конфигурация" у меня нет раздела MQTT
хотя на странице "Управление sonoff-rf pin" - MQTT  включен
в данный момент у меня прошито: 
spiffs.0xBB000_flash_size_1Mb.256Kb_2018.12.03 / build.0x00000_flash_size_1Mb.256Kb_2018.12.02

renat2985, 2020-01-28:
Возможно когда вы ставили галочку на MQTT, вы не перезагрузил esp8266 - это обязательно.

klepan63, 2020-01-28:
и чего я с ним только не делал
и как его только не перезагружал - и кнопкой и отключением от сети
но на странице "Конфигурация" раздел MQTT не появляется
что заметил - в файле /modules.json не появляется module mqtt
он же должен там появится после выставления галки на странице "Управление sonoff-rf pin" ?
так же вручную забил настройки MQTT в /config.save.json и загрузил через file manager - результата нет

almaz8811, 2020-04-05:
Тоже не появляется MQTT ни в настройках, ни в /modules.json

almaz8811, 2020-04-08:
Удалось вписать вручную модуль mqtt в modules.json, появились настройки mqtt, но они не сохраняются и нет подключения к брокеру.

artgareev, 2020-06-03:
Прошил с нуля, включил mqtt, перезагрузил, настройки тоже не появляются.

tretyakovsa, 2020-06-03:
@almaz8811 Очень может быть что MQTT не работает вообще в этой версии.
Так как тогда никто не смог предложить внятного описание зачем он нужен и как его лучше использовать. У меня были попытки делать это для Iot менеджера, потом эта программа долго была не доступна. А сейчас больше проблем с серверами.
Сейчас есть решение с IOT менеджером но без нормального веб интерфейса https://github.com/DmitryBorisenko33/esp32-esp8266_iot-manager_modules_firmware