tretyakovsa/Sonoff_WiFi_switch #178: Поломалась вся графика

stishka1, 2020-02-08:
Здравствуйте.
Что случилось с 2-мя разными модулями?
NodeMcu и witty cloud
Все нормально отображалось, а спустя некоторое время одновременно на обоих выдается такая картина. Доступ есть но графика поломалась. Прошивки одинаковые:
build.0x00000_flash_size_1Mb.256Kb_2018.12.02.bin
spiffs.0xBB000_flash_size_1Mb.256Kb_2018.12.03.bin

Платы не создали свою точку, перезагрузка не помогла.

[image]

stishka1, 2020-02-08:
Может дело в роутере?

renat2985, 2020-02-09:
Нужно больше информации, покажите Консоль.

[image]

[image]

stishka1, 2020-02-09:
[image]

Перекинуло по ошибке на эту страницу:
https://developer.mozilla.org/ru/docs/Web/JavaScript/Reference/Errors/Unexpected_type

stishka1, 2020-02-09:
У 2-го модуля в коде вылезла такая ошибка:

[image]

И перекинуло на: https://developer.mozilla.org/ru/docs/Web/JavaScript/Reference/Errors/Not_defined

renat2985, 2020-02-10:
Попробуйте сменить дизайн: http://192.168.1.70/skins?set=index.htm

stishka1, 2020-02-10:
это помогло. А в чем причина была?

renat2985, 2020-02-11:
Покажите этот ваш файл от 2 устройств: http://192.168.0.70/wifi.scan.json
Что-то не так именно с ним.

stishka1, 2020-02-16:
[image]

stishka1, 2020-02-20:
сдвигов нет? непонятно почему крашится?

stishka1, 2020-02-20:
в любой момент может все это бахнуться?

tretyakovsa, 2020-02-21:
@stishka1 
Думаю нужно скачать файл конфигурации и сценариев и в архиве передать мне.
Скорее всего модуль использует много памяти и библиотека ArduinoJson все рушит у меня есть решение для новой версии прошивки и ваши настройки были бы очень кстати для сборки новой версии.

stishka1, 2020-02-29:
не пойму как и откуда его качать?
вот что-то скачал
[scenary.save.txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/4270239/scenary.save.txt)