tretyakovsa/Sonoff_WiFi_switch #179: Приложение,аналог для управления устройствами

Bender229, 2020-02-15:
Дорогие разработчики,есть ли способ голосового управления для iphone,с помощью голосовых комманд,как это представлено для андройда? Для ваших устройств,может есть аналоги управления которые можно связать с вашим устройством(nodemcu)может кто из камюнити пытался,за ранее буду благодарен если такое существует)

tretyakovsa, 2020-02-16:
Без звуковых функций для iphone попробуйте это приложение: 
https://apps.apple.com/us/app/e-smart/id1463342455

Bender229, 2020-02-16:
> Без звуковых функций для iphone попробуйте это приложение:
> https://apps.apple.com/us/app/e-smart/id1463342455

Спасибо,да не,у данного приложения отклик достаточно долгий,я и вэб интерфейсом пользуюсь:)
Просто смотрю даже что без хабов айфон и ндэмку не подружить)

klepan63, 2020-02-18:
голосом можно управлять через Быстрые Команды (Shortcuts)